maddevsio/mad-location-manager #2: Sample not working

ZaydelEduard, 2018-03-24:
Where i can see documentation? Or working sample? For now, sample not detect location or anything, tested on Genymotion, and two real phones.

Lezh1k, 2018-03-25:
@ZaydelEduard  hello. 
Documentation is here : https://github.com/maddevsio/mad-location-manager/wiki/HowTo-use-library . 
Sample application called "sensor data collector" is in repo too.

ZaydelEduard, 2018-03-25:
Yep, i know about your sample application!
I cloned that, started - but as result - nothing.
No detected my real location, no tracked anything.

Lezh1k, 2018-03-25:
Ok then, could you check GPS settings? I mean GPS should be enabled somewhere in settings (usual settings->location or settings->privacy->location).

ZaydelEduard, 2018-03-25:
Yep, i turned on GPS on phone and Genymotion also. And have stable internet connection. Nothing changed

Lezh1k, 2018-03-25:
So GPS enabled and used as location source (GPS used, not Wi-fi and mobile network) and nothing is happening. That's really strange, but I hope we will find the reason. 
Ok, could you check if log file exists? It should be located on your external storage -> SendorDataCollector folder.  
If yes - can you provide it?

puzanov, 2018-03-30:
@ZaydelEduard ? any progress?

puzanov, 2018-04-02:
Closing this issue. Will open when @ZaydelEduard will return.

closed 2018-04-02 by puzanov