maddevsio/mad-location-manager #6: GPS Status Disabled

moti0375, 2018-04-28:
Hi,

From some reason, even when GPS is enabled, the GPSEnabledChanged callback initially gives a false status followed by a true value. 

Thanks,
Moti.

Lezh1k, 2018-05-02:
Sorry for late answer. 
Fix is on the way, today I'm going to release .

moti0375, 2018-05-02:
Thanks you man!

Regards,
Moti.

Lezh1k, 2018-05-07:
@moti0375 
Hello. Seem's like it's expected behavior. When service is started this callback should return true. When stopped - false (GPS should be enabled in settings).

moti0375, 2018-05-07:
Hi Thanks, 

So you mean I should use: `ServicesHelper.addLocationServiceStatusInterface(mLocationServiceStatusInterface);` only after ` service.start()`

Something like this:

`KalmanLocationService.Settings settings = new KalmanLocationService.Settings(Utils.ACCELEROMETER_DEFAULT_DEVIATION,
                Utils.GPS_MIN_DISTANCE,
                interval,
                Utils.GEOHASH_DEFAULT_PREC,
                Utils.GEOHASH_DEFAULT_MIN_POINT_COUNT,
                Utils.SENSOR_DEFAULT_FREQ_HZ,
                this, false);

        service.reset(settings); //warning!! here you can adjust your filter behavior
        service.start(); 
        ServicesHelper.addLocationServiceStatusInterface(mLocationServiceStatusInterface);
`

OK

Lezh1k, 2018-05-28:
Yes, you are right. :)

closed 2018-05-28 by Lezh1k