maddevsio/mad-location-manager #7: Gps min time bellow the GPS intervals

yanai, 2018-05-07:
Hi,
Thanks for this promising library - It looks great.

We started testing it and we found that we don't get location callbacks in high frequency (higher than the default updates of the GPS). So for example, if we set it to 200 ms we would expect to get 5 updates a sec, but we still got about 1 GPS a sec (our GPS device updates)

From what I understand, even if the GPS data is not changed with such high frequency, the accelerometer has high-frequency updates, so you can fuse the "old" GPS (provided about one sec ago) values with fresh accelerometer values. 

Am I right? 
Do you think its possible have high frequent callbacks from the Mad-Location Manager?

Lezh1k, 2018-05-08:
Hello. Thank you for your interst . It's possible to get coordinates more frequently, but sometimes it leads to big errors. At least huge error comes from accelerometer drift. So now callback based on gps+filter. But sure we can add setting for using accelerometer. And it works ok if you don't have gps conection issues

Lezh1k, 2018-05-08:
Let's add such setting. Something like useAccelerometerDataAsMainSource. It will take about 2 days, ok?

yanai, 2018-05-09:
Thanks. For for the late reply...
If the drift is so high, I'm not sure about the value of it...
So do you think there is a way to get better positions frequencies with more accurate data ?

Lezh1k, 2018-05-17:
@yanai  Sorry for this delay. 
Drift of accelerometer is high enough and without correction within 5 seconds values become very inaccurate. There is information in android documentation about accelerometer drift, but seems like I couldn't use it in right way. 

I think it's possible to get better positions frequencies with more accurate data after calibration step. But I couldn't calibrate accelerometer without some user actions.

Lezh1k, 2018-07-11:
Won't fix (very bad test results). 
Sorry about that.

closed 2018-07-11 by Lezh1k