maddevsio/mad-location-manager #8: Incorrect Speed readings

moti0375, 2018-05-08:
Hi Again,

I looks like that the speed readings are incorrect.
Today I drove with the new release 0.1.11 - and the speed readings were incorrect. 

Even when I stopped, it keep show some incorrect speed readings. 
on some point when I drove about 60-70 kmh it showed more than 100 mSec. 

Switching to Android GPS provider gives the correct speed. 

(in my app I've implemented both mad and android location providers and I can switch between them from settings in run time)

Thanks again,
Moti.

Lezh1k, 2018-05-08:
Hello. 
I'm not sure if we do anything with speed :) I'll check. Maybe there is some initial values or temporary filter results. 
Thank you for feedback, @moti0375

Lezh1k, 2018-05-17:
@moti0375  
Hello. Sorry for this delay, I'm gonna to fix this at Friday .

Lezh1k, 2018-07-11:
Fixed.

closed 2018-07-11 by Lezh1k

BravoJason, 2018-07-11:
@Lezh1k Hi, I didn't see any of update for this issue. If it is possible, could you please update the code?