maddevsio/mad-location-manager #9: Covariance matrix Q

RonitBustin, 2018-05-16:
Hi. Thanks for sharing the code of your project.
Have been looking through your blog and your code. Note that the matrix Q is a covariance matrix and as such must be symmetric. The specific values you have chosen seem good, meaning that the matrix should be the following:
Q_k = [ \sigma_posX^2            0         \sigma_posX*\sigma_velX           0             \\
               0            \sigma_posY^2               0      \sigma_posY*\sigma_velY           \\
               \sigma_posX*\sigma_velX           0         \sigma_velX^2             0            \\
               0        \sigma_posY*\sigma_velY            0     \sigma_velY^2            ]

Lezh1k, 2018-05-17:
Hello. Thank you for interesting in our project. 
You are definitely right about covariance matrix Q. 

We have this : 
```
double velDev = accDev * m_predictCount;
        double posDev = velDev * m_predictCount / 2;
        double covDev = velDev * posDev;

        double posSig = posDev * posDev;
        double velSig = velDev * velDev;

        double Q[] = {
                posSig, 0.0, covDev, 0.0,
                0.0, posSig, 0.0, covDev,
                0.0, 0.0, velSig, 0.0,
                0.0, 0.0, 0.0, velSig
        };
```
As you said, posSig should be posSigX and posSigY, but we can get speed measurements only from GPS receiver, so for both directions we have one deviation. That's why we use 1 value.

RonitBustin, 2018-05-17:
That's not what I meant. The off-diagonal values of Q are the cross covariance values. As everything is real in this scenario, the matrix must be symmetric (with respect tot he man diagonal). Thus the matrix should be:
double Q[] = {
                posSig, 0.0, covDev, 0.0,
                0.0, posSig, 0.0, covDev,
                covDev, 0.0, velSig, 0.0,
                0.0, covDev, 0.0, velSig
        };

with two additional covDev values instead of two of the zeros in the lower part of the matrix.

Lezh1k, 2018-05-17:
I see now ))) Will try it today :+1:

closed 2018-05-28 by Lezh1k

RonitBustin, 2018-05-31:
Hi,
Saw that you uploaded a new version a couple of days ago.
The matrix Q in GPSAccKalmanFilter.java is still non-symmetric. 
Wanted to know if there was a problem when one changes it to be symmetric? 
Thanks,
Ronit

Lezh1k, 2018-05-31:
Hello. 
I didn't upload anything yet. Version still 0.1.11 :)

RonitBustin, 2018-05-31:
OK. Thanks!