maddevsio/mad-location-manager #10: Manipulating R matrix

yanai, 2018-05-23:
Hi,
As we want to manipulate and see if we can gain more accurate results, we would like to manipulate posSigma and velSigma in the R matrix. This can be done by using a factor parameter on the settings (and even expose it on the settings of the app for calibration).

Can you expose the settings such factor parameters?

Lezh1k, 2018-05-24:
Hello. 
Do you need to change it in run-time or we just need to add such coefficient to settings and multiply R matrix by that coeff?

yanai, 2018-05-24:
Hi,
Best is to have it at runtime, as we are not sure how and what are the correct values in advance. Once you enable it, we can expose it in the app's (advanced) settings, so we can test and change it easily, and see the results.

What do you think ?

Lezh1k, 2018-05-24:
Ok. I will implement it, but I'm not sure about time, sorry. 
Will do my best on weekends.

yanai, 2018-05-29:
Hi - Any updates on this issue ?

Lezh1k, 2018-05-30:
Not yet, sorry about delays.

Lezh1k, 2025-06-13:
Finally fixed [here](https://github.com/maddevsio/mad-location-manager-lib)

closed 2025-06-13 by Lezh1k