maddevsio/mad-location-manager #12: Clarification on  orientation:

arsab, 2018-05-29:
Understanding is that 
X is in Longitude towards East
Y is in Latitude towards East

Reflected by expressions below:
can we confirm the expression here in file "KalmanLocationService.java"

  xVel = speed * Math.cos(course);
  yVel = speed * Math.sin(course);  

Typically course being angle from North to East should we rather read/write ?

   xVel = speed * Math.sin(course);
   yVel = speed * Math.cos(course);

what would be my misunderstanding?

Thanks.

     x = loc.getLongitude(); 
     y = loc.getLatitude();

m_kalmanFilter.update(
                    sdi.getTimestamp(),
                    Coordinates.longitudeToMeters(sdi.getGpsLon()),
                    Coordinates.latitudeToMeters(sdi.getGpsLat()),
                    xVel,
                    yVel,
                    sdi.getPosErr(),
                    sdi.getVelErr()
         );

m_kalmanFilter.predict
                 (sdi.getTimestamp(), 
                  sdi.getAbsEastAcc(), 
                  sdi.getAbsNorthAcc()
                 );

m_kalmanFilter = new GPSAccKalmanFilter(
                    false, //todo move to settings
                    Coordinates.longitudeToMeters(x),
                    Coordinates.latitudeToMeters(y),
                    xVel,
                    yVel,
                    m_settings.accelerationDeviation,
                    posDev,
                    timeStamp
                   );

Lezh1k, 2018-05-30:
Hello, thanks for your interesting and attention. 
As I know course (NMEA) is the angle between the north direction of the meridian at the place of measurement and the direction of the path is counted clockwise from the direction to the geographical north. In fact, it shows the direction of the ground speed relative to the north. So I don't think that there is mistake.

closed 2018-06-01 by Lezh1k

arsab, 2018-09-07:
Is not this same issue as?
https://github.com/maddevsio/mad-location-manager/issues/24