maddevsio/mad-location-manager #16: Работа сервиса, когда приложение в фоне

ViTVetal, 2018-07-06:
Здравствуйте!
Крутая библиотека и как раз то, что мне надо. Возник вопрос:

Для отслеживания водителя я использую foreground service. Делаю это, так как обычный фоновый service имеет ограничение на количество запросов местоположения в новых версиях Android + вариант с foreground сервисом выглядит более стабильным (не хотелось бы иметь дело с ситуациями, когда приложение не посчитало "пробег" такси из-за убитого в фоне сервиса). 
Как лучше использовать ваше решение , что бы избежать подобных ситуаций (когда, к примеру, для использования навигации водитель свернет приложение)?

Lezh1k, 2018-07-09:
Здравствуйте. 
Спасибо за проявленный интерес. 

По поводу работы в фоне - мы наткнулись на очень большое количество подводных камней и странностей, связанных с сервисами. Поэтому текущее решение не рекомендуется использовать в фоне долгое время.
 
Например : в телефонах xiomi при включенном режиме энергосбережения сервис всё равно может быть выгружен из памяти, даже если используется PowerManager.WakeLock. Если же добавить приложение в исключения (есть там такая возможность), то всё равно через некоторое время что-то происходит с timestamp-ом всех событий. 

В описанной Вами ситуации (водитель стал сверяться с картой, к примеру, или ему кто-то позвонил и приложение свернулось) обычный сервис, как в текущем решении, отрабатывает нормально. Думаю foreground сервис будет работать ещё стабильней.

ViTVetal, 2018-07-11:
Возможно имеет смысл, в будущих версиях, дать возможность пользователям запустить сервис как background так и foreground?

Lezh1k, 2018-07-11:
Думаю лучше вообще отказаться от сервисов, а сделать просто возможность подавать на вход координаты/данные датчиков, а на выходе получать уже сглаженные данные. И оставить возможность пользователю самому решать, что он хочет использовать - сервис или foreground сервис или как-то по другому.

closed 2018-07-13 by Lezh1k