maddevsio/mad-location-manager #17: Question for get absolute acceleration.

BravoJason, 2018-07-11:
I have a question for get absolute acceleration in KalmanLocationService class.

In onSensorChanged callback function:
`
android.opengl.Matrix.multiplyMV(absAcceleration, 0, rotationMatrixInv, 0, linearAcceleration, 0);
`
The question is that I think we need to do rotate from the device coordinate to global coordinate and we also need to do projection in each global coordinate axis.

But for our code, I only see the rotation code and I didn't see the projection code.

Could you tell me is there any misunderstanding of the code?

BravoJason, 2018-07-11:
You are right.

When we use the rotation matrix to rotate the vector, it will rotate the coordinate system.

It needn't to do projection.

closed 2018-07-11 by BravoJason

Lezh1k, 2018-07-11:
https://blog.maddevs.io/reduce-gps-data-error-on-android-with-kalman-filter-and-accelerometer-43594faed19c 

Part with "Determining orientation of the phone in space" explains that. 
Also greater explanation is here : https://www.youtube.com/watch?v=T9jXoG0QYIA&t=675s - ~11 min.