maddevsio/mad-location-manager #23: Adaptability of code

tony98789, 2018-08-22:
Hi, just a question not an issue but was not sure where the appropriate place to ask was. So i have two question regarding the implementation of your kalman filter. 

Firstly, how adaptable is this. Currently i have an android app that posts it location to a web api at regular intervals. I want to filter the data at server level to keep a log of the original data. I use the fused location provider in my app. So in theory could the kalman filter be moved to a stand alone library that i could just post the gps and sensory data too? Completely new to signal noise and filtering so sorry if these questions sound silly.

And secondly how is the battery consumption with combining the sensory data? Using the fused location provider i have had no battery issues. So does your approach reduce the battery life faster?

Again apologies if this is not the correct place to be asking these questions.

Lezh1k, 2018-08-23:
Hello, @tony98789 
Thank you for your interest :) 

1. You can easily port this solution to server side. There are 2 examples : C++/Java, so you can choose what's better for you. You just need to provide GPS/sensors data with timestamps. And you need to make some queue for these events.  

2. About battery consumption. Main battery consumer is GPS receiver. And I was able to reduce GPS calls frequency to 1 call / 8 seconds without quality loose. But it depends on phone and sensors you have.  So  answer to your second question - battery won't be an issue in case of using this approach. 

This is right place for questions :)

tony98789, 2018-08-23:
@Lezh1k Brilliant thank you for your reply. Your explanation of how the filtering works was great, best source of information i have found on this topic. I will most likely port it to c# though since i have .net api and the java is not far from c# anyway. Thanks again :).

closed 2018-08-23 by tony98789

puzanov, 2018-08-23:
@tony98789 porting is always welcoming :) Feel free to ask questions, will be happy to help.