maddevsio/mad-location-manager #24: Incorrect speed longitude and latitude in KalmanLocationService.OnLocationChanged

SHuygh, 2018-08-29:
If I am not mistaken, a few small errors are present in the calculation of the velocities and the correction of the absolute accelerations.

Currently they are implement as follows
        xVel = speed * Math.cos(course);
        yVel = speed * Math.sin(course);
As xVel represents the speed from West to East and the Cos(0) is equal to 1, you would get the the speed is completely projected on xVel when traveling North and yVel would be 0.
        xVel = speed * Math.**sin**(course);
        yVel = speed * Math.**cos**(course);

Furthermore in SensorGpsDataItem the absolute accelerations are corrected as follows
        this.absNorthAcc = absNorthAcc * Math.cos(declination) + absEastAcc * Math.sin(declination);
        this.absEastAcc = absEastAcc * Math.cos(declination) - absNorthAcc * Math.sin(declination);
The magnetic declination is defined negative if the magnetic north is to the "west" of the true north, which results in the signs being opposite of what the are now.
        this.absNorthAcc = absNorthAcc * Math.cos(declination) **-** absEastAcc * Math.sin(declination);
        this.absEastAcc = absEastAcc * Math.cos(declination) **+** absNorthAcc * Math.sin(declination);

Lezh1k, 2018-08-31:
Hello, @SHuygh 
I'll check again, thank you. 
Maybe I should just swap X and Y. :) Seems like there is wrong aliases for directions.

arsab, 2018-09-07:
is not this same issue as?
https://github.com/maddevsio/mad-location-manager/issues/12#issuecomment-393027538

Lezh1k, 2018-09-10:
Thank you, @arsab . Seems you are right

closed 2018-10-12 by Lezh1k

arsab, 2019-11-06:
Understand now why this error was not visible.
Actually it was not computed because Kalman filter update does not take into account speed:
Because measurement dimension is 2 (no speed)