maddevsio/mad-location-manager #25: Using a list of gps points.

tony98789, 2018-10-20:
Hi,

So your combining sensor and location data. Would it still be possible to do this without the sensor data. For example the predict function. Lets say we have a list of Locations. We Initialise the Kalman Filter with the first Location. And then loop through each location, on each iteration we call predict and then update, we then use the predicted values to update and return the location object. I assume this would reduce the accuracy, but this should still reduce noise and jumping correct? 

I ask because have been implementing the kalman in c#. And i only have the Location data, so no sensor data to work with. On the predict function could i not just use the xVal and yVal like below.

`var xVal = loc.Speed * Math.Sin(loc.Bearing);
 var yVal = loc.Speed * Math.Cos(loc.Bearing);`

loc being the android Location object. I am still learning how the whole kalman fits together so sorry if i am way off with this question.

Lezh1k, 2018-10-21:
Hi @tony98789 :) 
You can use [this](https://github.com/andreynovikov/GeoTrackFilter) solution. 
Also you can try current solution with some modifications...You need to change Q matrix at least :)

tony98789, 2018-10-23:
@Lezh1k Thank you, that is a great help. In regards to the noise. So When i instantiate the GeoTrackFilter i set the noise to 1.0. I then loop through each location object and call UpdateVelocity. I have the accuracy for each location returned form the android device. Would i need to update the noise on each iteration with the accuracy i have?

Lezh1k, 2018-10-24:
@tony98789 
Yes, you need to update noise values.

closed 2018-11-12 by Lezh1k