maddevsio/mad-location-manager #26: Could not resolve library path

Gabriel0402, 2018-10-27:
Built new app with the configure error:
Could not find method compile() for arguments [com.github.maddevsio:mad-location-manager:0.1.9] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.

Lezh1k, 2018-10-28:
Hello. Try to change build file. Actual version of library is 13

Lezh1k, 2018-10-28:
0.1.13 I mean

Gabriel0402, 2018-10-28:
Still got the error: Could not find method compile() for arguments [com.github.maddevsio:mad-location-manager:0.1.13] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.

Here's the build file:
buildscript {
    
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.1.3'
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        google()
        jcenter()
        maven { url 'https://jitpack.io' }
    }
}

dependencies {
    compile 'com.github.maddevsio:mad-location-manager:0.1.13'
}

But could build by importing the library:
settings: include ':app',':madlocationmanager'
build: api project(path: ':madlocationmanager')

Lezh1k, 2018-10-29:
Try replace "compile" with "implementation". Maybe you have newest version of android studio and gradle . (compile is deprecated as I know)

Lezh1k, 2018-11-12:
@Gabriel0402 any progress here? )))

Gabriel0402, 2018-11-12:
tried implementation, still the same error here.

Lezh1k, 2018-11-12:
What's your environment? I mean java version, studio version, OS etc. :) 
I just can't reproduce so I think it's somehow related to that stuff

Gabriel0402, 2018-11-12:
I'm using java 8, 
Android Studio 3.1.3
Build #AI-173.4819257, built on June 4, 2018
JRE: 1.8.0_152-release-1024-b01 x86_64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
and I tried both mac and windows 10, neither works.

Lezh1k, 2018-11-12:
Ok, let me try to reproduce it on mac.

Lezh1k, 2019-06-20:
can't reproduce.

closed 2019-06-20 by Lezh1k