maddevsio/mad-location-manager #28: callback error

bomeeraa, 2018-12-14:
[image]

MainActivity.java
private void initActivity() { } 
ServicesHelper.getLocationService(this, value -> {}

context = MainActivity
callback = {-$lambda@---}

class com.example.lezh1k.sensordatacollector.-$$Lambda$MainActivity$TCctZCsoVok8NZ14_qqYhiCfNqQ

i guess missing MainActivity.

please help me

----
[image]

Lezh1k, 2018-12-17:
Hello @bomeeraa 
Is this sample project? 
Can you provide more details about this issue?
Thank you.

Lezh1k, 2019-06-20:
Closing due to inactivity

closed 2019-06-20 by Lezh1k