maddevsio/mad-location-manager #29: cannot be cast to mad.location.manager.lib.Services.KalmanLocationService$LocalBinder

MosCriogen, 2019-01-17:
[image]

Спасибо за огромный труд! Ваша библиотека - моя последняя надежда на успех предприятия, которое уже год как не может быть завершено из-за того что телефоны врут. Особенно когда лежат на месте и никуда не едут.

Я подключил библиотеку как указано в описании. Подключил внутри сервиса который implements GoogleApiClient.ConnectionCallbacks, GoogleApiClient.OnConnectionFailedListener, LocationListener

потому что это то единственное решение которое мне позволило обойти DOZE mode. Мне нужны геоданные пока экран отключен и даже если телефон не трогали пол-часа. Оказалось он не засыпает только  когда службы гугла хотят координат тоже.

Помогите пожалуйста

MosCriogen, 2019-01-17:
[image]

MosCriogen, 2019-01-17:
У данной проблемы ноги растут из файла манифеста. Решено.

Секция, отвечающая за службу из которой я запускаю KalmanLocationService:
       <service
            android:name="com.orderssupportapp.TestLocationUpdates"
            android:icon="@mipmap/ic_launcher"
            android:enabled="true"
            android:exported="true"
            android:launchMode="singleInstance"
            android:process=":service">
        </service>

Вот это дело - android:process=":service" надо прописать в секцию отвечающую за KalmanLocationService:

       <service
            android:name="mad.location.manager.lib.Services.KalmanLocationService"

            android:process=":service"

            android:enabled="true"
            android:exported="false"
            android:stopWithTask="false" />

closed 2019-01-17 by MosCriogen

Lezh1k, 2019-01-18:
Спасибо за информацию.

MosCriogen, 2019-01-18:
А можно спросить у Вас совет, может есть более приличный способ как выключить DOZE?