maddevsio/mad-location-manager #30: Poor documentation

erlanamanatov, 2019-01-28:
Hi, I am working with this library, it works fine with default parameters, but I would like to know more about the  settings, so  I have some questions:
1)  there are 2 not described parameters for KalmanLocationService.Settings (VelFactor, PosFactor). What they are for? How to use them?
2)  what is the radius of “merging area” for GeohashRTFilter? For example, if the hash length is 8 and the point count is 2.
3)  GeohashRTFFilter calculates 4 distances (distanceGeoFiltered, distanceGeoFilteredHP, distanceAsIs, distanceAsIsHp). What is what? Which one should I use?
4) How is  IsRunning() method (KalmanLocationService) supposed to work? It returns false even when my app is getting location updates, is it ok?

Lezh1k, 2019-01-31:
Hello :) 

1. Those parameters are used in R matrix as gain coefficients. If you know that your velocity (from gps receiver) is (let's say 2 times) more reliable than position you can set those parameters like 0.5 for velocity and 1.0 for position for example. 

2. For geohash with precision 8 radius is ~±0.019km. Point count is used for point validation. For example if you have received points_count sequenced points with same geohash - this point is valid. If you have received less then points_count - this point isn't valid yet. 

3. 
DistanceGeoFiltered - distance filtered with geohash filter. 
DistanceAsIs - uses all received points. 

HP suffix means high precision. That means that for distance between two points we use more accurate algorithm. 

4. Seems like it's bug. It should return false when it isn't started :) 

Thank you for your interest.

erlanamanatov, 2019-01-31:
I have tested the GeohashRTFilter with different parameters, I think the radius of "merging area" (for hash length = 8) is much less than 19 m. In my app 2 points at a distance of 5 meters were not merged. 
[image]

I have also tested the filter with parameters of 8-2 and 7-2  (hash length and min point count). Results are not acceptable, too many points were lost because of driving speed

Lezh1k, 2019-02-01:
Sorry. 7, 2 for cars.

Lezh1k, 2019-06-20:
Update is on the way.

closed 2019-06-20 by Lezh1k