maddevsio/mad-location-manager #33: Location not getting correctly.

adhithyagnair, 2019-02-05:
Hi, 

I'm trying to develop a driver tracking application. At first I was using just Android's LocationManager API with GPS as the only provider. The output was not so perfect, it had jumps all over my route. 

It was then I tried using this library. But I lost a lot of locations after that.

[image]

In this picture Red line shows my route towards destination and Green line shows my route back to where I started.  And you can clearly see from the picture that only a few point are plotted. But out of my surprise I got the exact point from where I took the U turn. 

NB: The vehicle was continuously moving throughout the route.

My system requires a foreground notification and I'm writing my code on that service. Here is what I included in the code:
`public class LocationService extends Service implements LocationServiceInterface {
    .......

    private GeohashRTFilter m_geoHashRTFilter;

    @Override
    public void onCreate() {
        super.onCreate();
m_geoHashRTFilter = new GeohashRTFilter(Utils.GEOHASH_DEFAULT_PREC, Utils.GEOHASH_DEFAULT_MIN_POINT_COUNT);
         startForeground(KEY_NOTIFICATION_ID, getForegroundNotification);
    }

    @Override
    public int onStartCommand(Intent intent, int flags, int startId) {

        ServicesHelper.addLocationServiceInterface(this);
        ServicesHelper.getLocationService(this, value -> {
            if (value.IsRunning()) {
                return;
            }
            m_geoHashRTFilter.stop();
            value.stop();
            KalmanLocationService.Settings settings =
                    new KalmanLocationService.Settings(Utils.ACCELEROMETER_DEFAULT_DEVIATION,
                            Utils.GPS_MIN_DISTANCE,
                            Utils.GPS_MIN_TIME,
                            Utils.GEOHASH_DEFAULT_PREC,
                            Utils.GEOHASH_DEFAULT_MIN_POINT_COUNT,
                            Utils.SENSOR_DEFAULT_FREQ_HZ,
                            null, false, Utils.DEFAULT_VEL_FACTOR, Utils.DEFAULT_VEL_FACTOR);
            value.reset(settings); 
            value.start();
        });
        return START_STICKY;
    }

@Override
    public void locationChanged(Location location) {
        if (location != null ) {
                DataManager.getInstance(LocationService.this)
                        .saveDeviceTracking(location.getLatitude(), location.getLongitude());
            }
   }
...........
}`

Please help me if I'm missing anything in this.

Lezh1k, 2019-02-05:
Hello. :) 

Please try to set geohash default precision to 6 and geohash default min point count to 2.

adhithyagnair, 2019-02-05:
Ok. I'll do it. And I have an other doubt, From the documentation and sample I found that we can filter the obtained location using 
`m_geoHashRTFilter.filter(location);` 

And  the filtered location from it by executing the following code.
`   List<LatLng> routeFilteredWithGeoHash = new ArrayList<>(m_geoHashRTFilter.getGeoFilteredTrack().size());

        for (Location loc : new ArrayList<>(m_geoHashRTFilter.getGeoFilteredTrack())) {
            routeFilteredWithGeoHash.add(new LatLng(loc.getLatitude(), loc.getLongitude()));
        }`

And I'm expecting the last location that I fed for filtering will be available as the last entry in this list (list here: routeFilteredWithGeoHash ). Is that correct? 

Or there is anything more to it?

erlanamanatov, 2019-02-05:
As far as I can see, you are saving KalmanFiltered locations in locationChanged() method, not GeohashRTFilter filtered. So I don't think, that you are losing a lot of data because of geohash precision and geohash min point count parameters.

adhithyagnair, 2019-02-06:
I have tried both of them, for geoHashing along with the above mentioned code I have updated my locationChanged() method as 
`  if (location != null ) {
            m_geoHashRTFilter.filter(location);
            List<LatLng> filteredLocation = getFiltered Location();
            if (!filteredLocation.isEmpty()) {
                int len = filteredLocation.size();
                DataManager.getInstance(LocationBgService.this)
                        .saveDeviceTracking(filteredLocation.get(len - 1).getLatitude(), filteredLocation.get(len - 1).getLongitude());
            }
        }`

` public List<LatLng> getRoute() {
        List<LatLng> routeFilteredWithGeoHash = new ArrayList<>(m_geoHashRTFilter.getGeoFilteredTrack().size());

        for (Location loc : new ArrayList<>(m_geoHashRTFilter.getGeoFilteredTrack())) {
            routeFilteredWithGeoHash.add(new LatLng(loc.getLatitude(), loc.getLongitude()));
        }
        return routeFilteredWithGeoHash;
    }`

where LatLng is a POJO that I have created.

But even though the result is not so trust worthy.

Lezh1k, 2019-02-07:
I still can't understand what's happening. 
Do you receive some wrong coordinates (like on this https://user-images.githubusercontent.com/30405024/52267760-e7da8300-295f-11e9-95b9-448b402939fa.png screen) ? 
Or you don't receive wrong coordinates, but the route contains small amount of coordinates ? 

Sorry, can't get it. 
Also have you tried our example application on this route?

adhithyagnair, 2019-02-07:
At first without using geohash I got only a few but [correct](https://user-images.githubusercontent.com/30405024/52258131-ed29d480-2943-11e9-997f-fa671953371a.png) coordinates. 

Later I have used geohashing with precision to 6 and min point count to 2. 
It has few more [correct](https://user-images.githubusercontent.com/30405024/52267756-e5782900-295f-11e9-872b-516af3487b41.png) points compared to the first one, but surprisingly I'm getting some points from [wrong](https://user-images.githubusercontent.com/30405024/52267760-e7da8300-295f-11e9-95b9-448b402939fa.png) coordinates :(

Lezh1k, 2019-02-07:
Now I got it. )) Thanks.

Lezh1k, 2019-02-07:
Seems you've got (0, 0) coordinate? I'll look at this today :) 
What about our demo application ?

adhithyagnair, 2019-02-08:
No, it is not (0,0) coordinate. I couldn't test with your demo app, I'll check it anyway.

Lezh1k, 2019-02-11:
I mean that there is (0, 0) somewhere compensated by Kalman filter. 
Can you share your project? I'll try to reproduce :)

closed 2019-06-20 by Lezh1k

thxmxx, 2019-08-29:
Check your GPS frequency I bet it's 1Hz or bellow and you are not having enough points GEOHASH_DEFAULT_MIN_POINT_COUNT for the precision GEOHASH_DEFAULT_PREC.

Try walking instead of driving a car.... lol

farfromrefug, 2019-09-20:
@Lezh1k about the (0,0) coordinates. I was getting those too. For me the reason was that i was using an emulator without sensors. So i was only getting sdi from ```onLocationChange``` and so the KalmanFilter was never "initialized" with a Sensor event. I fixed it as explained here https://github.com/maddevsio/mad-location-manager/issues/52#issuecomment-533188474