maddevsio/mad-location-manager #35: Library usage

powerzaba, 2019-02-16:
Hi! I'm trying to make an android application that should rely on high accuracy GPS data, but the default GPS raw data have a lot of noise in them, so after a bit of researching I discovered that I could use the phone's sensors together with the kalman filter in order to filter out the noise! And then I stumbled upon your library! I'm a bit of a novice when it comes to android development and I'm having a hard time trying to understand how to use the tools that you provided! I've looked into your example application but I still couldn't figure out how to use it. For my example I don't really need to display the position on a map, but I need to translate the GPS coordinates into an imaginary grid. Now my question is, how do I transform the sensors data and the GPS data into inputs for a kalman filter? (I believe you're already doing, but I can't understand how). And if I can convert them into proper inputs could I only use your KalmanFilter.java class?

Lezh1k, 2019-02-18:
Hello, @powerzaba 

There is "H" matrix in kalman filter for inputs transformation. In my example it's identity matrix (with ones on main diagonal) because all transformations I do before filter. I convert GPS coordinates into distance between north pole and current GPS location. Then I convert sensors data to IMU (inertial measurement unit). And then I can use both of them in Kalman filter. 

About library usage - you can follow next steps (described here https://github.com/maddevsio/mad-location-manager/issues/22) :
1. Create new class that implements LocationServiceInterface. 
2. Register it with this :
```
settings = new KalmanLocationService.Settings(Utils.ACCELEROMETER_DEFAULT_DEVIATION,
                    Utils.GPS_MIN_DISTANCE,
                    Utils.GPS_MIN_TIME,
                    Utils.GEOHASH_DEFAULT_PREC,
                    Utils.GEOHASH_DEFAULT_MIN_POINT_COUNT,
                    Utils.SENSOR_DEFAULT_FREQ_HZ,
                    null, false, Utils.DEFAULT_VEL_FACTOR, Utils.DEFAULT_POS_FACTOR);

    ServicesHelper.addLocationServiceInterface(this);
```
But change default geohash precision and default geohash min point count. 
3. Implement this:
```
@Override
public void locationChanged(Location loc) {
}
```

That's it. You should receive location in locationChanged method .

powerzaba, 2019-02-18:
Hi! I've looked at the thread that you suggested me to look into and I've tried to use the code that the user MosCriogen posted, but nothing happens! I don't even get the permission request from Android to allow the use of location/GPS, here's the simple code I'm using. (Basically the same from MosCriogen)

`package com.example.kalmangpsfilter; 

import android.content.Context;
import android.location.Location;
import android.widget.TextView;

import mad.location.manager.lib.Commons.Utils;
import mad.location.manager.lib.Interfaces.LocationServiceInterface;
import mad.location.manager.lib.Services.KalmanLocationService;
import mad.location.manager.lib.Services.ServicesHelper;

public class Kalman implements LocationServiceInterface {
    private Context m_context;
    private KalmanLocationService.Settings settings;
    private TextView locationText;

    public Kalman (Context context, TextView textView) {

        m_context = context;
        locationText = textView;

        settings = new KalmanLocationService.Settings(Utils.ACCELEROMETER_DEFAULT_DEVIATION,
                Utils.GPS_MIN_DISTANCE,
                Utils.GPS_MIN_TIME,
                Utils.GEOHASH_DEFAULT_PREC,
                Utils.GEOHASH_DEFAULT_MIN_POINT_COUNT,
                Utils.SENSOR_DEFAULT_FREQ_HZ,
                null, false, Utils.DEFAULT_VEL_FACTOR, Utils.DEFAULT_POS_FACTOR);

        ServicesHelper.addLocationServiceInterface(this);
    }

    public void start_k () {

        ServicesHelper.getLocationService(m_context, value -> {
            if (value.IsRunning()) {
                return;
            }

            value.stop();
            value.reset(settings); //warning!! here you can adjust your filter behavior

            value.start();
        });
    }

    public void stop_k () {

        ServicesHelper.getLocationService(m_context, value -> {
            value.stop();
        });
    }

    @Override
    public void locationChanged(Location location) {
        locationText.setText("Long: "+location.getLongitude()+"Lat: "+location.getLatitude());
    }
}

And in my main i have this:
` package com.example.kalmangpsfilter;

import android.support.v7.app.AppCompatActivity;
import android.os.Bundle;
import android.widget.TextView;

public class MainActivity extends AppCompatActivity {
    private TextView location;
    private Kalman kalman;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        location = findViewById(R.id.location);
        kalman = new Kalman(this, location);
    }

    @Override
    public void onStart() {
        super.onStart();
        kalman.start_k();
    }
}
`

So I'm trying to log the filtered coordinates into a simple text field, but nothing happens, what am I doing to wrong? (The function locationChanged never occurs);

powerzaba, 2019-02-18:
Update! I managed to make it work and log the coordinates! But the update frequency it's too low, I tried to change the settings in the setting object, but I get updates like every minute.

Lezh1k, 2019-02-19:
@powerzaba  Try to log every GPS coordinates received by https://github.com/maddevsio/mad-location-manager/blob/master/app/src/main/java/com/example/lezh1k/sensordatacollector/Loggers/GPSDataLogger.java, please. 

I faced this situation when tested application inside building, but I don't think that it's your situation.  Let me know about GPSDataLogger, please. :)

powerzaba, 2019-02-19:
Hi! I cannot get the LocationManager to pass to the GPSDataLogger, so I'm unable to log, or should I create a new LocationManager?

Lezh1k, 2019-02-20:
Add own GPS listener in this case and compare receiving frequencies from standard gps listener with kalmanlocationservice, please. 

Also try to disable geo hash filter by setting zeroes to it's settings.

Lezh1k, 2019-06-20:
Closing due to inactivity

closed 2019-06-20 by Lezh1k