maddevsio/mad-location-manager #36: How i use Kalmanlocation inside my own service

mindfulritik, 2019-02-18:
ServicesHelper.addLocationServiceInterface(SendLocation.this);
when I call this function oncreate of Service class then I get 
 
java.lang.ClassCastException: android.os.BinderProxy cannot be cast to mad.location.manager.lib.Services.KalmanLocationService$LocalBinder

Servicehelper lineno:67

MosCriogen, 2019-02-18:
https://github.com/maddevsio/mad-location-manager/issues/29

the same - problem is in manifest

MosCriogen, 2019-02-18:
find this string:   android:process=":service" in manifest 's section with  your service and copy to section with lib

MosCriogen, 2019-02-18:
example:

my own service with calman lib calls
<service
        android:name="com.orderssupportapp.TestLocationUpdates"
        android:icon="@mipmap/ic_launcher"
        android:enabled="true"
        android:exported="true"
        android:launchMode="singleInstance"
        android:process=":service">
    </service> 

lib section:
   <service
        android:name="mad.location.manager.lib.Services.KalmanLocationService"

        android:process=":service"

        android:enabled="true"
        android:exported="false"
        android:stopWithTask="false" />

Lezh1k, 2019-02-19:
@MosCriogen  thank you

mindfulritik, 2019-02-19:
thank you @MosCriogen, now my problem solved .

closed 2019-02-19 by mindfulritik