maddevsio/mad-location-manager #37: Java Control Matrix Column Number Bug?

aflofo, 2019-03-06:
I by means am no expert in this field, but am implementing my own version of this library and found that the control matrix here: [https://github.com/maddevsio/mad-location-manager/blob/master/madlocationmanager/src/main/java/mad/location/manager/lib/Filters/GPSAccKalmanFilter.java](https://github.com/maddevsio/mad-location-manager/blob/master/madlocationmanager/src/main/java/mad/location/manager/lib/Filters/GPSAccKalmanFilter.java) was ``m_kf = new KalmanFilter(4, mesDim, 1);`` when I beleive the control matrix should be 2 columns not 1 otherwise the matrix multiplication wouldn't work and also the B control matrix from your blog was two columns. It is also 2 in the cpp version of the GPSKalmanFilter. Kinda confused how this even worked to begin with because you technically should have gotten an indexing error. But anyways those are my findings.
Cheers,
@aflofo

Lezh1k, 2019-03-06:
Hello, @aflofo 
Control matrix will have 2 columns: 
```
        this.B = new Matrix(stateDimension, controlDimension);
        this.Uk = new Matrix(controlDimension, 1);
```
But control vector will always have 1 column. 

Thank you for interest.

aflofo, 2019-03-06:
Correct so shouldn’t controlDimension be 2 instead of 1

Lezh1k, 2019-03-06:
Yes, seems like there is mistake and it should be 2 instead of 1 in last arg. Thank you :)

Lezh1k, 2019-06-20:
Fixed

closed 2019-06-20 by Lezh1k