maddevsio/mad-location-manager #38: Example app is not working

anton-nagornyi, 2019-03-13:
I made a build of the SensorDataCollector example app. Then I tried to run it on a 4 different Android phones. It worked only on one of them. By saying "worked" I mean it started to track route. Do you have any idea what is wrong with other devices? Your help is very appreciated ))

Lezh1k, 2019-03-18:
Hello, @anton-nagornyi 

1. Check if gps location is enabled
2. Check if motion sensors are available
3. Provide info about whole bunch of phones, please :)

anton-nagornyi, 2019-03-18:
Hello, @Lezh1k 

1. Gps is enabled
2. I guess you are right, there is no gyroscope on those devices at least. Do we need that?
3. Please take a look at Meizu M3 note and Bravis A504 Trace.

Thank you!

makarenkovov, 2019-03-18:
I confirm. Here are 2 more models. And judging by the debugger, the coordinates are obtained and processed, but the track is not displayed. Honor 7A DUA-L22, samsung Galaxy j3(2016) SM-J320F

makarenkovov, 2019-03-19:
on Xiaomi Redmi Note 5 (it's with gyroscope) its'work. Without gyroscope TYPE_LINEAR_ACCELERATION not working.

What means "It's possible to use just TYPE_ACCELEROMETER with high-pass filter."?

Lezh1k, 2019-03-21:
Maybe it's because of disabled gyroscope... 

It's possible to use just TYPE_ACCELEROMETER with high-pass filter means that it's possible to get linear acceleration (related to x/y/z axis) using result of high-pass filter. But it doesn't give best result. 
More information could be found here  - https://developer.android.com/reference/android/hardware/SensorEvent.html#values

Lezh1k, 2019-06-20:
closing due to inactivity.

closed 2019-06-20 by Lezh1k