maddevsio/mad-location-manager #39: Any ideas - how else to improve the track?

makarenkovov, 2019-03-19:
Get rid of the stars finally?
[image]
[image]
Here such parameters were used:
public static final int GEOHASH_DEFAULT_PREC = 8;
public static final int GEOHASH_DEFAULT_MIN_POINT_COUNT = 3;

now added (while testing):
public static final int GPS_MIN_DISTANCE = 20;

Lezh1k, 2019-03-20:
Try geohash precision = 7 and min point count = 3 :)

makarenkovov, 2019-03-20:
Now trying geohash precision = 7 and min point count = 2

( it's wrong combination of settings (geohash - nothing remains for geohash )) ):
public static final int GEOHASH_DEFAULT_PREC = 8;
public static final int GEOHASH_DEFAULT_MIN_POINT_COUNT = 3;
public static final int GPS_MIN_DISTANCE = 20;)

Lezh1k, 2019-03-20:
GPS_MIN_DISTANCE - this value for your gps receiver. 
GEOHASH_* - geohash settings. 
All these value should be provided into service (reset method).

makarenkovov, 2019-03-20:
На русском изьяснюсь, чтобы понятнее было, что хотел сказать...
Да я все это понял - что для чего и куда. Мой предыдущий пост про то, что с такой настройкой геохешинг все отбросит, т.к. точки получаем через каждые 20 м, а геохешингом пытаемся отфильтровать 3 точки в радиусе 19 метров )) - что у меня и произошло)

Lezh1k, 2019-03-21:
В этом случае да, геохэш всё отбросит. 
Вероятно в этом случае нужно взять геохэш с бОльшим радиусом, но тогда траектории становятся такими... более смазанными и усредненными. Можно отключить геохэш вообще, но тогда надо очень тщательно подобрать коэффициенты в матрицы шумов. Обычно для каждого телефона/транспорта разные.

Lezh1k, 2019-06-20:
Закрываю, вроде вопрос исчерпан. Активности нет.

closed 2019-06-20 by Lezh1k

makarenkovov, 2019-06-21:
I can add. To reduce the jump. It is necessary to leave only GPS in the settings of the phone source.