maddevsio/mad-location-manager #40: compiling Visualizer + disabling GeoHash

zobis2, 2019-03-27:
heya , really nice project ! 
few questions  👍 
1. 
how do i compile the visualizer in order to play the log files(ubuntu or windows) ? i haven't found any Makefile in the project itself.
2.
 regarding GeoHash , if i want to disable it i just need to give 0,0 to his config under utils?
if so the green line on map will be only the raw GPS and the blue line will be the Correct GPS coordinate after kalman filter with corrected accelerometer  values at that time frame right?

my green and blue lines are on the same coordinates although ive tried to walk few mins in an open streen with the application .

thanks alot.

Lezh1k, 2019-03-28:
Hello. Thank you for interest. 

1. It's qt-based project, so you need to use qmake. It's much better and more comfortable to use qtcreator. 
2. Yes, you need to set 0,0 .

wurongfu1992, 2019-04-04:
Yes,my green and blue lines are on the same coordinates, I have try several times; It`s my phone sensor problem or I need change the configure？

ReonLion, 2019-04-22:
> Yes,my green and blue lines are on the same coordinates, I have try several times; It`s my phone sensor problem or I need change the configure？

Hello, did you figure out what caused green and blue lines are on the same coordinates?

wurongfu1992, 2019-04-22:
> > Yes,my green and blue lines are on the same coordinates, I have try several times; It`s my phone sensor problem or I need change the configure？
> 
> Hello, did you figure out what caused green and blue lines are on the same coordinates?

sry，I don't; I debug the original location info and kalman location info, find they are almost the same, except for accuracy.

closed 2019-06-20 by Lezh1k