maddevsio/mad-location-manager #42: Map tracking is not working for me !

DevAhmed4, 2019-04-26:
I have downloaded the Sensor data collector app but it doesn't seem to work with me.
by "Not working" i mean it doesn't track the route !
[image]

My mobile is infinix Note 4 and i checked that it has an accelerometer & gyroscope . my GPS is working and has the permission ,  my network is ON , and this is what my log file looks like !

[image]
 i am appreciating any help!

Lezh1k, 2019-04-29:
Hello. 
Seems like you don't receive gps coordinates (In peace of log that you've provided). 
Could you send full log and check that GPS used as main source of locations ?

Lezh1k, 2019-06-20:
Closing due to inactivity

closed 2019-06-20 by Lezh1k