maddevsio/mad-location-manager #45: How does this library account for sensors with incompatible update intervals?

edorphy, 2019-07-18:
A couple of my cheap Android dev phones don't give me any sensors back so I can't step through the code to trace and understand. Does this implementation work for sensors with different update intervals? I.e. if the accelerometer is 10 hz but gps is only 1 hz, can the predict be called more than once before the update is called? (GPS update interval is limited by AOSP or no reception like in a tunnel)

Conversely, if the accelerometer is subject to too much noise and values are unstable, can the update be called more than once before a single predict is called? (Running with phone and accelerometer isn't smooth)

Lezh1k, 2019-07-18:
>Does this implementation work for sensors with different update intervals? 
Yes, it's kind of hack here because we use several predict steps and one update step (when GPS coordinate received)

edorphy, 2019-07-19:
Could you point me to the code where the magic is? I didn't see any accumulation or averaging to account for multiple predicts prior to an update--but could have missed something. Thanks!

Lezh1k, 2019-07-19:
Sure. 
There is predict method after each reception of rotation vector data, right? 
And there is update method after each reception of GPS data. 
Sensor data frequency is ~50Hz, but GPS frequency much smaller. 

Code could be little bit confusing, but there it is:
```
if (sdi.getGpsLat() == SensorGpsDataItem.NOT_INITIALIZED) {
                        handlePredict(sdi);
                    } else {
                        handleUpdate(sdi);
                        Location loc = locationAfterUpdateStep(sdi);
                        publishProgress(loc);
                    }
```

closed 2019-07-22 by Lezh1k