maddevsio/mad-location-manager #49: Проверить есть ли необходимые сенсоры для работы библиотеки

soulsb, 2019-08-28:
Я хотел у себя в приложении убедиться что у телефона есть все необходимые сенсоры для работы библиотеки, и если нету то выполнить другой код...

Я проверяю на сенсоры TYPE_LINEAR_ACCELERATION , и TYPE_ROTATION_VECTOR, пример кода ниже.

И так все нормально определялось, до того момента как в руки попал смартфон Galaxy A5 2017, он проверку на эти два сенсора проходит, но location не получаю, и я начал подозревать, может еще на какие то сенсоры надо проверку делать, какие еще сенсоры требует библиотека??

```
SensorManager sensorManager = (SensorManager) getSystemService(Context.SENSOR_SERVICE);
            Sensor sensor = sensorManager.getDefaultSensor(Sensor.TYPE_LINEAR_ACCELERATION);
            Sensor sensor2 = sensorManager.getDefaultSensor(Sensor.TYPE_ROTATION_VECTOR);
            
            Intent intent;
            if (sensor == null || sensor2 == null)
            {
                // если какой то сенсор не обнаружен, то выполняю другой код
            }
            else {
                intent = new Intent(this, MadLocationMonitoringService.class);
            }
            startService(this,intent);
```

Lezh1k, 2019-08-28:
Нет, другие сенсоры не нужны.
Проверьте, что GPS включен и все необходимые разрешения есть.

soulsb, 2019-09-02:
да GPS включен и все необходимые разрешения есть...

понятно, значит будем разбираться. Спасибо

Lezh1k, 2019-09-02:
Встречался с таким поведением, когда забыл включить GPS. 
Можете проверить, что просто GPS координаты от обычного LocationListener-а приходят? В каком -нибудь тестовом приложении.

puzanov, 2019-09-02:
@soulsb 
Возможно на телефоне не откалиброван GPS, не найдено нужное количество спутников

soulsb, 2019-09-02:
@puzanov  да, так и вышло, оказалось что надо было откалибровать датчики. После калибровки все заработало.

@Lezh1k  @puzanov  спасибо за помощь 👍

closed 2019-09-02 by soulsb