maddevsio/mad-location-manager #50: Speed values and Q matrix

thxmxx, 2019-09-02:
Can someone tell me why rebuildQ is in function of m_predictCount since the accDev does not change over time?

Also, do you have some clue on why the speed readings are so wrong?

Thank you

Pstoppani, 2019-09-02:
I’ve also been working on fixing the speed reading and this is what I came up with (in C++):

void GPSAccKalmanFilter::rebuildQ(double dtUpdate, double accDev) {
    double velDev = accDev * dtUpdate * 100;
    double posDev = velDev * dtUpdate * 100;
    double covDev = velDev * posDev;
    double posSig = posDev * posDev;
    double velSig = velDev * velDev;
    
    m_kf.Q.setData(16,
                   posSig, 0.0, covDev, 0.0,
                   0.0, posSig, 0.0, covDev,
                   covDev, 0.0, velSig, 0.0,
                   0.0, covDev, 0.0, velSig);
}

thxmxx, 2019-09-02:
@Pstoppani I'm also using a similar approach:

    private void rebuildQ(double dtUpdate, double accDev) {
            double velDev = accDev * dtUpdate;
            double posDev = (accDev * dtUpdate * dtUpdate) / 2.0;
            double covDev = velDev * posDev;
    
            double posSig = posDev * posDev;
            double velSig = velDev * velDev;
    
            double Q[] = {
                    posSig, 0.0, covDev, 0.0,
                    0.0, posSig, 0.0, covDev,
                    covDev, 0.0, velSig, 0.0,
                    0.0, covDev, 0.0, velSig
            };
            m_kf.Q.setData(Q);
        }

I'm working on android btw, there was also a wrong size of B matrix and u vector. They should be (4x2) and (2x1) respectively.

Btw, now it seems the vectors (x,y) from accelerometer are not on the same orientation as the gps speed vectors.

thxmxx, 2019-09-02:
I think it's also a mistake to use the orientation from the gps on update cause we have this data more accurate from magnetometer.

Pstoppani, 2019-09-02:
Agreed.  I did the same thing.

farfromrefug, 2019-09-16:
@thxmxx @Pstoppani can you elaborate on what you changed for gps orientation?

Pstoppani, 2019-09-16:
Sorry, I misunderstood the previous comment.  I think it is fine to use the GPS orientation on update since the accel “predicts” are from the IMU at a higher rate than GPS.  The Kalman filter combines them properly and I think it is best to use both GPS and IMU orientation.

farfromrefug, 2019-09-17:
@Pstoppani thanks! Will test it on my side. I ported the code to JS as i use it within Nativescript cross platform environment. 
Thanks

farfromrefug, 2019-09-24:
@thxmxx thanks a lot! will use that too. 
Now did you get the speed to be accurate? That s actually the reason why i am looking at this lib. 
My client find the GPS speed to not be accurate and responsive enough. So i look here.
Now i have to lib working correctly except for the speed. The reading are wrong.
I have looked at both rebuildQ implementations here but i dont get a correct result.
Well it see wrong result with my JS port. Need to test android version with your implementation to see what i get

thxmxx, 2019-09-24:
@farfromrefug you should also use this #51 
It's already merged btw

thxmxx, 2019-09-24:
> Now did you get the speed to be accurate? 

I'm getting the error of +-5Km/h. But I'm still working on that, since I'm using GPS speed on updates and the predict results are still bad.

farfromrefug, 2019-09-25:
@thxmxx Thanks for the infos!
I added the bearing from sensors as you mentioned (Also reported in test app for easier debug). You can see it here :
https://github.com/farfromrefug/mad-location-manager
Now what i am seeing is a lot more accurate speed reading from the GPS right now. 

My question is this. When updating the Kalman filter with the GPS location, what we are doing is using GPS speed as a reference for the Kalman Filter to "predict" it afterwards based on Linear ACceleration, right? I am not that familiar with the Kalman Filter
But if it is so shouldn't we "reset" the Q matrix on new Location update?

What i am seeing when outside is that the GPS speed pretty quickly goes to 0 when i stop walking. But the Filtered speed does not want to go to 0. It just slowly goes down.
Also Do you use the ```m_useGpsSpeed``` of the GPSKalmanFilter? I am not sure what it does.
Thanks

closed 2025-06-13 by Lezh1k