maddevsio/mad-location-manager #57: Results worst after filtering with kalman.

klagupl, 2019-10-31:


Lezh1k, 2019-10-31:
Approach is definitely correct. 
But need more info about your setup and filter settings.

klagupl, 2019-10-31:
I use only the kalman filter implementation from this repo. So basicaly i initialized filter with first point data, then I predict with acceleration, and update with gps data.
No additional settings. 
Maybe the problem is, I use wrong standard deviation for acceleration and position. 
Or the accelerometer is not calibrated?

Lezh1k, 2019-10-31:
Maybe you use uncalibrated accelerometer. 
There are 2 variants of using filter here : 
Use speed readings from GPS receiver or not use. 

Also this sensor bosch bno055 - did you calibrate it? I mean it uses magnetometer and when you use magnetometer you should compensate soft and hard iron distortions there .

Lezh1k, 2019-10-31:
That's why I asked about your setup :)

klagupl, 2019-10-31:
I'm using variant with speed from GPS. The magnetometer is calibrated according to my mechanical compass. 
Maybe frequency is the issue? It's only 5hz.

Lezh1k, 2019-10-31:
No, it should not be an issue.

klagupl, 2019-11-04:
So here is what i found, looking at the path. 
[image]

On the image above we can clearly see that the path is not correct, it should by along a street. 
And when the filter is updated with correct gps data, it's ignored and getting even worse.
So the filter is not filtering at all. Prediction works, but updating not.
Any ideas ? :D

Userpc1010, 2020-12-25:
Sensor is bosch bno055 it has a drawback, it cannot accurately maintain orientation during prolonged accelerations and vibrations, these are problems of its fusion algorithm, it may be worth trying a more reliable sensor.

Lezh1k, 2025-06-13:
There were too many errors in this implementation. Probably later I'll fix them, but now I'd recommend to use [new version](https://github.com/maddevsio/mad-location-manager-lib)