maddevsio/mad-location-manager #61: Tracking information is not displayed

altshuler, 2019-11-23:
Hello.
I'm downloaded and installed the application and tried it on 2 different devices -
Galaxy S9+ and Galaxy S6. On both devices the map and tracking information
is not displayed. First times after installation application is crashing.
Here attached link to the log file:
https://drive.google.com/open?id=1Vrv1iQGsjBQTB-uWQl9ddoXtyipWcu4a

dmitofclubs, 2019-12-08:
Note your http request is `401` that means "Not autorized". 
Indeed, mapbox api [access token](https://github.com/maddevsio/mad-location-manager/blob/master/app/build.gradle) from this repo is invalid, If you try to load map using it, you will get nothing: http://jsfiddle.net/k7Lnh9t6/

You should register own account on mapbox.com and get your public accees token. Or take it from anywhere.

I had the same issue. But now with my own access token I have got this stupid error (app running through Android Studio):

```
I/System.out: [CDS][DNS] getAllByNameImpl netId = 0
D/libc-netbsd: [getaddrinfo]: mtk hostname=api.mapbox.com; servname=(null); netid=0; mark=0
    getaddrinfo( app_uid:10171
D/libc-netbsd: getaddrinfo() uid prop:
    getaddrinfo() getuid():10171
    [getaddrinfo]: mtk ai_addrlen=0; ai_canonname=(null); ai_flags=4; ai_family=0
D/libc-netbsd: [getaddrinfo]: mtk hostname=api.mapbox.com; servname=(null); netid=0; mark=0
D/libc-netbsd: getaddrinfo( app_uid:10171
    getaddrinfo() uid prop:
    getaddrinfo() getuid():10171
    [getaddrinfo]: mtk ai_addrlen=0; ai_canonname=(null); ai_flags=1024; ai_family=0
D/libc-netbsd: getaddrinfo: api.mapbox.com get result from proxy gai_error = 0
I/System.out: [CDS]rx timeout:10000
E/NetworkUtilities: [MTK_net]ss.ss_family:10
I/System.out: [socket][5] connection ********;LocalPort=59811(10000)
E/NetworkUtilities: [MTK_net]ss.ss_family:2
I/System.out: [CDS]connect[api.mapbox.com/13.33.244.9:443] tm:10
D/Posix: [Posix_connect Debug]Process com.example.lezh1k.sensordatacollector :443 
E/NetworkUtilities: [MTK_net]ss.ss_family:2
E/NetworkUtilities: [MTK_net]ss.ss_family:2
I/System.out: [socket][6:59811] exception
E/NetworkUtilities: [MTK_net]ss.ss_family:2
I/System.out: close [socket][********]
I/Mbgl-HttpRequest: Request failed due to a connection error: Failed to connect to api.mapbox.com/13.33.244.9:443 
E/Mbgl: {ordatacollector}[Setup]: loading style failed: Failed to connect to api.mapbox.com/13.33.244.9:443
```

But `api.mapbox.com` is 13.33.244.9! I don't know how and where fix this typo.

[image]

altshuler, 2019-12-08:
Hi, 
my problem is solved.
Just changed the token to valid one and changed 
**compileSdkVersion**  to   **28**  in  build.gradle  file.

dmitofclubs, 2019-12-17:
When I connected to internet via mobile operator (rather than through wi-fi), then app become to working. You can close this issue.

FX-HAO, 2020-04-24:
I know this's irrelative, but could someone tell me how to show device location and get to my current location on Mapbox?

Lezh1k, 2020-04-25:
Isn't it implemented in demo application, @FX-HAO  ? 
I don't remember all details, so, try this :  m_map.setMyLocationEnabled(true); 
Hope it will help