maddevsio/mad-location-manager #65: Not working on Android API 22

dziry, 2020-04-20:
**Description:** I've tested the example "Sensor Data Collector" on both Android API 29 and API 25. It works perfectly.

But for Android API 22 it seems not workig as it should be : Calibration is always 0.000000% when trying to calibrate and when starting to track nothing shows on the map.

**Expected behavior:**
- Click on Start Calibration : Progression from 0 to 100%
- Click on Start Tracking : Shows the current location and one or more routes

**Source code:**
- `getCompletePercentage()` always returns 0.0
- `locationChanged(Location location)` never triggered

**Android API version:** 22

**Device:** Emulator.

dziry, 2020-04-20:
Nvm, I figured out reading the issues #11 and #38 that problem is probably happening because the sensors `TYPE_LINEAR_ACCELERATION` and `TYPE_ROTATION_VECTOR` are not available.

So, i will reformulate my question : 
1. Does anyone tried to solve this using other sensors like the accelerometer or the magnetometer in order to detarminate acceleration and oriontation?
2. Otherwise, could you give us some big guidlines (classes/functions/lines to edit) so we can improve your library in order to handle this case.
3. How much time this task may take?
4. Is it worth it (accuracy)? 

Thank you

Lezh1k, 2020-04-21:
Hello, @dziry  . 
Thank you for interest in our library. Unfortunately it's kind of rusty now and need to be refactored. So I'll just try to answer your questions above :) So: 

1. Yes, I've tried to solve this using raw accelerometer, gyroscope and magnetometer data. I've used [Madgwick filter](https://x-io.co.uk/open-source-imu-and-ahrs-algorithms/). It works perfectly, but you need to know, that on some devices X,Y and Z axis are rotated. I mean, that when I tested results from ROTATION_VECTOR_SENSOR and results from Madgwick filter I saw, that Y == X and X == -Y . Maybe this is related to wrong implementation of that filter, but I tried on raw sensors (with STM32) and there was no error. So you need to keep in mind, that this is possible. 

2. To improve this I was planning to do next things:
- Make some ISensorDataCollector interface with 2 methods: Start() and Stop(). All sensor data providers should implement that interface. 
- Make some interface IOrientationProvider.  This one should return quaternion with current orientation. 
- Implement several IOrientationProviders, based on ROTATION_VECTOR, Madgwick filter, alpha-beta filter and, maybe, something else (some custom implementation of sensor fusion with Kalman filter, for example). 
- Then this one should be as data source for Kalman filter. I don't think that we need to improve Kalman filter, but I wanted to try to implement extended Kalman filter and see if it's better or not. 

3. I don't think that it will take much time. I know, that this code isn't perfect and should be refactored and redesigned. So, in current implementation it will take from 1 to 2 weeks, I guess. 

4. I'm pretty sure, that Madgwick filter works fine in our case. The main problem we will have on smartphones without Magnetometer. I don't know how to deal with it right now. 

Hope it will help you. Also I hope that I'll get more free time and return to this project. Thank you one more time.

closed 2025-06-13 by Lezh1k