maddevsio/mad-location-manager #66: Why does R get larger as accuracy gets larger?

FX-HAO, 2020-04-24:
As far as I know, the theory is that if the measurement noise (accuracy) is small, the measurement is trusted more and R should get smaller and go to 0. But the following code seems to do the opposite.
https://github.com/maddevsio/mad-location-manager/blob/7a8ca33705cd79a00e9647f0eb40e3c3e73b451e/madlocationmanager/src/main/java/mad/location/manager/lib/Services/KalmanLocationService.java#L564
Then `posDev` is assigned to `posSigma` and is used here: 
https://github.com/maddevsio/mad-location-manager/blob/7a8ca33705cd79a00e9647f0eb40e3c3e73b451e/madlocationmanager/src/main/java/mad/location/manager/lib/Filters/GPSAccKalmanFilter.java#L84-L89

Lezh1k, 2020-04-24:
Hello @FX-HAO 

https://developer.android.com/reference/android/location/Location#getAccuracy()

getAccuracy returns some radius in meters. If it's bigger then accuracy of coordinate is smaller. 
Hope it will help to understand :)

FX-HAO, 2020-04-25:
So if the accuracy is bigger, the measurement noise is bigger, then R should be smaller. 
Should accuracy of coordinate be the denominator like the following?
```java
 double R[] = { 
         1/posSigma, 0.0, 0.0, 0.0, 
         0.0, 1/posSigma, 0.0, 0.0, 
         0.0, 0.0, 1/velSigma, 0.0, 
         0.0, 0.0, 0.0, 1/velSigma 
 }; 
```

Lezh1k, 2020-04-25:
Yes. But getAccuracy() result becomes bigger when measurement noise becomes bigger. So R should be bigger too in this case.

FX-HAO, 2020-04-26:
Yes, you're right, R gets bigger in this case. Sorry for the confusion.

closed 2020-04-26 by FX-HAO