maddevsio/mad-location-manager #71: can we add a network provider in case of GPS is unavailable?

FX-HAO, 2020-05-12:
Since GPS is accurate enough, can we add a network provider to improve accuracy in case of GPS is unavailable?

FX-HAO, 2020-05-12:
We can fuse location information from many different sources (such as GPS and Wi-Fi), does it help to improve accuracy? I guess it's what [FusedLocationProvider](https://developers.google.com/location-context/fused-location-provider) does?

Lezh1k, 2020-05-12:
FusedLocationProvider does all info, you don't need to fuse sensor data with that thing :)

FX-HAO, 2020-05-12:
But most of our phones (in china) don't have google play services installed :(

So I can just add this line and nothing else need to be done, right?
```
m_locationManager.requestLocationUpdates(LocationManager.NETWORK_PROVIDER, m_settings.gpsMinTime, m_settings.gpsMinDistance, this);
```

Lezh1k, 2020-05-13:
I don't think that LocationManager.NETWORK_PROVIDER works with enough precision, sorry. 
So if you don't have google play services installed, then you need to try our approach. But you need GPS at least + set of sensors (accelerometer, gyroscope, magnetometer) .

FX-HAO, 2020-05-13:
Yes, location information from network provider has less accuracy. But I'm thinking of a case that GPS is unavailable but the network works (I am not sure whether or not this could happen). In this case we are losing all location information without network provider. 
And I think Kalman filter can correct these locations from network. Since their accuracy is bigger, the measurement is trusted more.

Lezh1k, 2020-05-13:
It will work, I guess.
But you need to know, that noise integration produces very big error. So you can get very inaccurate position in this case.

FX-HAO, 2020-05-13:
Thanks for the suggestion! I'll give it a try and feed back :)