maddevsio/mad-location-manager #85: Facing issue in building the Project in Android Studio

dharmendra08, 2020-09-22:
I built the project in Android studio and it is showing some error. 

Function Name - KalmanLocationService.Settings

```
constructor Settings in class Settings cannot be applied to given types;
required: double,int,int,int,int,int,double,ILogger,boolean,boolean,double,double
found: double,int,int,int,int,int,double,MainActivity,boolean,double,double
reason: actual and formal argument lists differ in length
```

Showing this error. Please help.

Lezh1k, 2020-09-22:
Hello. 
Can you share your call code? 

1. MainActivity -> should implement ILogger (should have method log or something like that)
2. You missed one boolean variable. 

Best regards, Oleg

dharmendra08, 2020-09-22:
Thanks, @Lezh1k for the speedy response. 

I am able to resolve this issue now and build the application. 

I had to add a boolean variable for `onlyGpsSensor`.

closed 2020-09-22 by dharmendra08