maddevsio/mad-location-manager #86: Accuracy Benchmarking - How?

dharmendra08, 2020-10-05:
Really liked the project and the support the team is providing. 

Currently, I am going through the algorithm and have built the APK for testing where I am collecting the data while driving through Bike and later analyzing the logged files. 

I want to know - How can I compare the accuracy with respect to normal GPS points? 

I have attached the log file of one of the trips I did near my locality. 

Settings used during this trip: 

> 1. GPS Min Time - 100ms
> 2. GPS Min Distance - 0m
> 3. Position Min Time - 100ms
> 4. Sensor Frequency - 50Hz
> 5. GeoHash Precision - 6
> 6. GeoHash Min Point - 3

[2020-09-29_2.txt](https://github.com/maddevsio/mad-location-manager/files/5326451/2020-09-29_2.txt)

Lezh1k, 2020-10-08:
Hello. 
Yeah, that was really tricky to measure accuracy. 
We used 2 approaches:
1. Generate ideal route and then add some noise (+/- random(1.5-50) meters to each coordinate. Also you need to generate accelerometer values between each coordinate and apply noise there too. 
After that apply our filter (with different parameters). Result route should be close to your generated route. All sharp jums should be smoothed. 
This will show how to work with filter at all + easy to visualize. But won't give you real result, because it's hard to generate good noise for accelerometer values.

2. Take 5-7 GPS loggers (I used smartphones) and walk around. Collect all logs and apply filter. 
Result route should be close to your route. It's better to use several devices (different models at least). 

Then apply filter and see results.

dharmendra08, 2020-10-12:
Thanks for the update. 

The current logging file generates GPS Points & Kalman Update Points. Where can I get the Geohash points in logging file?

georgiecasey, 2020-11-15:
@dharmendra08 Geohash points are prefixed with `FKS` in the logging file

closed 2020-12-10 by Lezh1k