maddevsio/mad-location-manager #92: jitpack dependency wrong

johannesneitzel, 2020-10-30:
When following the jitpack link stated in your readme.md it gets you here: https://jitpack.io/#maddevsio/mad-location-manager
Now 

 	dependencies {
 	        implementation 'com.github.maddevsio:mad-location-manager:Tag'
 	}

didn't work for me. I had to state the dependency to your lib like this:

 	dependencies {
 	        implementation 'com.github.maddevsio:mad-location-manager:0.1.13'
 	}

In order to avoid compability issues it's good practise to explicitily state the version of the external dependency.

> Caution: When specifying dependencies, you should not use dynamic version numbers, such as 'com.android.tools.build:gradle:3.+'. Using this feature can cause unexpected version updates and difficulty resolving version differences.

https://developer.android.com/studio/build/dependencies

Lezh1k, 2020-11-04:
@johannesneitzel 

Hello. Thank you. 
So  what do I need to change exactly? Maybe you can make pull request with these changes?

johannesneitzel, 2020-11-13:
@Lezh1k 

Thanks for getting back to me! The problem was that I didn't click on: 
[image]

Which then will generate the correct link! Thus this isn't an issue.

Lezh1k, 2020-11-15:
Ok, then I'll close this issue :)

closed 2020-11-15 by Lezh1k