maddevsio/mad-location-manager #98: Can use filter for drone?

Userpc1010, 2020-12-25:
This Kalmon filter algorithm can help improve the accuracy of GPS data on the drone, for example, fill in the gaps between GPS updates 5 to 90Hz? If use data from Boch BNO080 linear acceleration and quaternion absolute orientation resistant to acceleration and vibration on an aircraft, can expect an increase in accuracy GPS position?

Lezh1k, 2021-01-01:
Hello there :). 
Sure, it will imrove accuracy of GPS data. 
Any I believe you can also get sigmas from documentation to your accelerometer and update model.
But maybe your dron's flight controller already has good characteristics?

Userpc1010, 2021-01-01:
I intended to use a filter for a DIY flight controller. Sigmas if I understand correctly this is a characteristic of the accelerometer's accuracy? If yes, then the accuracy of linear acceleration is more dependent on vibration and calibration sing the error of subtraction the gravity vector of acceleration data is higher than own noise accelerometer, maybe it makes sense to leave it as before?

Lezh1k, 2021-01-09:
Sigma is a standard deviation of accelerometer. You can calculate it yourself, but very often you can find this information in manual.

Userpc1010, 2021-01-14:
Is it possible to use the horizontal position accuracy estimate (UBX nav-pvt) for the filter rebuildR (k, posDev) if  translate it into meters?

Lezh1k, 2021-01-15:
I guess you need another model for this.

Lezh1k, 2025-06-13:
Probably now it's easier to integrate it into drone. Please check [this version](https://github.com/maddevsio/mad-location-manager-lib)