maddevsio/mad-location-manager #100: xVel, yVel in handleUpdate

afracxas, 2021-01-05:
Hi, this is TOM.
I found some strange thing below code in handleUpdate function. It may be bug or not.
            double xVel = sdi.getSpeed() * Math.cos(sdi.getCourse());
            double yVel = sdi.getSpeed() * Math.sin(sdi.getCourse());
sdi.getCourse() return bearing of android location.
Unit of bearing is degree not radian. But argument of Math.cos/sin is an angle, in radians.
And the start point (zero degree) of bearing is north. so I think that code should be modified like below
            double xVel = sdi.getSpeed() * Math.sin(Math.toRadians(sdi.getCourse()));
            double yVel = sdi.getSpeed() * Math.cos(Math.toRadians(sdi.getCourse()));

Lezh1k, 2021-01-08:
Hello, Tom (@afracxas ) 

Thank you, I'll check this.

einjunge99, 2025-05-14:
Any updates on this?