maddevsio/mad-location-manager #103: KalmanLocationService crashing

pedromalta, 2021-01-27:
Whenever I start it I get a crash

`    java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$3.done(AsyncTask.java:354)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
        at java.util.concurrent.FutureTask.run(FutureTask.java:271)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:764)
     Caused by: java.lang.AssertionError
        at mad.location.manager.lib.Commons.Matrix.setData(Matrix.java:19)
        at mad.location.manager.lib.Filters.GPSAccKalmanFilter.rebuildB(GPSAccKalmanFilter.java:69)
        at mad.location.manager.lib.Filters.GPSAccKalmanFilter.predict(GPSAccKalmanFilter.java:124)
        at mad.location.manager.lib.Services.KalmanLocationService$SensorDataEventLoopTask.handlePredict(KalmanLocationService.java:250)
        at mad.location.manager.lib.Services.KalmanLocationService$SensorDataEventLoopTask.doInBackground(KalmanLocationService.java:323)
        at android.os.AsyncTask$2.call(AsyncTask.java:333)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:764) `

Any idea where to begin debuging this?

pedromalta, 2021-01-28:
It happens when I update gradle plugin from:
`classpath 'com.android.tools.build:gradle:4.0.1'`
to
`classpath 'com.android.tools.build:gradle:4.1.2'`

Lezh1k, 2021-01-30:
Seems like wrong matrix initialization or update. 
Could you provide more info? At least I need GPSAccKalmanFilters constructor. 
Did you try to debug it and see what's wrong with matrix dimensions?

smirnovegorv, 2021-02-01:
Hi. I encountered same issue.
Matrix dimensions are 4 rows and 1 column, while args.length = 8. This is a third call to Matrix.setData() after app is started.

This method is called from GPSAccKalmanFilter.rebuildB() method.
b argument is 
```
0 = 4.914112499999999
1 = 0.0
2 = 0.0
3 = 4.914112499999999
4 = 3.135
5 = 0.0
6 = 0.0
7 = 3.135
```

this.m_kf.B is empty 4x1 matrix.

I see it is created in KalmanFilter class. Which is created in GPSAccKalmanFilter with hardcoded stateDimensions = 4 and controlDimensions = 1. Which is inconsistent with b in rebuildB() method, which is always 8 elements long.

smirnovegorv, 2021-02-01:
Hmm, I see now, in your repository controlDimensions is already fixed to be 2. While the version 0.1.13 in jitpack has this issue. Please update your jitpack version if possible as it is pretty much outdated

smirnovegorv, 2021-02-01:
There is another similar crash in matrix.setData() caused by inconsistence between 4 parameters passed to m_kf.Zk.setData() on GPSAccKalmanFilter. Because 4 params are passed to the update matrix (2 coords and 2 velocities) but only 2 are expected (since useGpsSpeed is hardcoded to false in KalmanLocationService.java:614 which results in 2x1 matrix being created).

smirnovegorv, 2021-02-01:
Created a pull-request with a small fix for that

Lezh1k, 2021-02-02:
@smirnovegorv 
Thank you :) Project right now is kind of "rusty". 
But most fixes are moved to lib branch to C library.

closed 2021-02-02 by Lezh1k