maddevsio/mad-location-manager #109: Sensor calibration needed ?

ademar-prothon, 2021-06-17:
Hi ! 
I'm making a small project for my internship in a tech company, the goal is to locate and track a vulnerable road user to provide more security and i came across your library wich interested me very much.

I've implemented the KalmanLocationService in my own app but i'm still getting abnormal speeds when the precision is quite good. E.g i get 20 m/s in speed and weird bearing measures. 
I looked at the provided app and saw a lot of messages in the logcat, and there the precision was astonishing and i even got acceleration values, which i could be interested in getting too. 
My question is, do i need to calibrate the KalmanLocationProvider in my own app and in which way ? I browsed through the code and wasnt able to find how does the Sensor and Deviation calculator interacts with the provider service. I would also be interested in talking about the acceleration values, are they based on terrestrial referential or mobile one ?

I would be really pleased to get news from you, 

Adémar

Lezh1k, 2021-06-21:
Hello @ademar-prothon 
Thank you for interesting in the project. Unfortunately it requires more maintenance and debugging . At least we have an issue related to measurement units (as I understand we need to change seconds to microseconds at least. So it might be better to use ideas of this project, but not the implementation. 

 As regards to your question about calibrating - yes, you need to calibrate your sensors first.

Best regards, 
Oleg.

ademar-prothon, 2021-06-23:
Hi ! Thanks for your feedback !
Could you please give me a hint about how the class SensorCalibrator interacts or can do so with the KalmanLocationProvider ?
I looked through all the code and cant find any indication about how the calculated sigmas are used in the Kalman process.

Kind regards,

Lezh1k, 2021-06-27:
Those sigmas are used in R and Q matrices.

closed 2025-06-13 by Lezh1k