maddevsio/mad-location-manager #110: gpsacckalmanfilter tutorial video ?

imenemedjaoui, 2021-07-14:
Hello. I'd like to thank you first for this amazing library. I'd like to use it for my graduation project but I don't know how to. is there any tutorial video how to use the gps acceleration kalman filter fusion ? I think your library is the only thing that can save my project. thank you in advance.

Lezh1k, 2021-07-15:
Hello there. 
We will return to this library in 2 weeks (99% of confidence) . 
There are several issues with the library (for example hack with Q matrix, using rotation matrices instead of quaternions and so on) , but literally 30 minutes ago we have found resources to solve them.

And fixing them will take from 4 to 6 weeks I guess, so ETA - middle of the september. 
According to your question about video - no, we haven't any video, sorry. 
After those fixes we will update our readme and wiki here and record such video.

Lezh1k, 2025-06-13:
Yep, 2 weeks passed. Definitely. 
All fixes are in [new version](https://github.com/maddevsio/mad-location-manager-lib)

closed 2025-06-13 by Lezh1k