maddevsio/mad-location-manager #127: Old device without sensors TYPE_ROTATION_VECTOR, TYPE_LINEAR_ACCELERATION

daniel-costaa, 2023-05-04:
Hello! Before anything, I would like to thank you for all the ideas and effort put into this library.

I'm trying to adapt it for use on relatively old devices, and I saw that you left a comment saying that using just the `TYPE_ACCELEROMETER` sensor with a high pass filter should work.

I would like to know if you have any working example with this approach, because I have been trying to implement it for a few days, but without any success.

Lezh1k, 2023-05-04:
Hello @daniel-costaa 

Unfortunately I did not save any example related to high pass filters. 

Here are some ideas that might be helpful. 
1. If you don't have TYPE_ROTATION_VECTOR - you can use madgwick filter. [here](https://github.com/maddevsio/mad-location-manager/blob/lib_cpp/lib/src/MadgwickAHRS.cpp) is an implementation.  Also take a look at lib_cpp branch - it can be helpful too. There are implementations of quaternion, vector rotation and madgwick filter.  
2. If you don't have TYPE_LINEAR_ACCELERATION - please see [this](https://developer.android.com/guide/topics/sensors/sensors_motion#java) . There is an example of how to implement high-pass filter.

Lezh1k, 2023-05-05:
I believe with only accelerometer (without magnetometer) it's not possible to calculate position of phone and impossible to connect accelerometer data to GPS data.

daniel-costaa, 2023-05-05:
Got it, thanks for the help!

closed 2023-05-05 by daniel-costaa