maddevsio/mad-location-manager #128: GPSAccKalmanFilter

vinithlakshmanan, 2023-07-08:
Hi, Thanks for making the effort for code. I saw in your MainActivity Class only the GeoHashRTFilter is used? Have you used the GPSAccKalmanFilter class instead ? To see the difference in results?

closed 2023-07-09 by vinithlakshmanan

Lezh1k, 2023-07-10:
Hello there . 
Sure we used GPSAccKalmanFilter :) GeoHashRTFilter is kind of experiment or additional check (to filter obviously wrong coordinates and jumps).

vinithlakshmanan, 2023-07-15:
Hi Oleg,

Thanks for your reply. I was wrong. You indeed use the kalman filters.
I managed to get the app working on my Android phone ( pixel 6a ). However, my usage was slightly different from yours. I wanted to have speed at a high frequency.
Therefore, I forked your project and modified it to publish the speed after every update and predict step and then display it on the UI.  I did it after every update and predict, because the update is step is carried out roughly at 1hz ( the gps receiver) and the predict step using the IMU is done at a much higher frequency.

I tested out the app in the past days in a car. However, I found the values to be quite off. May I know if you have any experience on this or have you tested this before? If so, it would be great to have some feedback from you on where I can look to fix it.

Regards,
Vinith