maddevsio/mad-location-manager #130: Kalman location service started location random coords

isamirkhaan1, 2023-10-05:
Hey, 

I am using the Kalman Service to draw a route, in parallel, I'm also using the fucedlocation to draw another route. The purpose is to compare both paths and see which is more accurate.
For the last few hours, I've been getting weird random locations from the Kalman Service, which can be seen below.

[image]

Here is the implementation

```
internal class MadLocationManager(
    private val context: Context
) : LocationServiceInterface {
    private lateinit var geoHashRTFilter: GeohashRTFilter

    init {
        ServicesHelper.addLocationServiceInterface(this)
    }

    override fun locationChanged(location: Location?) {
        location ?: return
        Log.i(TAG, "locationChanged: $location")

            RxBus.send(location)
        }
  
}

```

Lezh1k, 2023-11-03:
Hello there. 
Actually there is an error in java version now. 
I recommend to try out version in lib_cpp branch (lib folder). But there is only C++ version of filter for now (with a quite simple API) .

gfrancischini, 2024-08-14:
Hi @Lezh1k. We are experiencing (maybe) a similar issue. Most of our users does not seems to have this error but on couple of Samsung devices we see wrong coordinates in the middle of nowhere (like in the middle of the ocean)

Do. you think that it is related to the bug mentioned in your comment? is there a new version with this bug fixed?

Lezh1k, 2025-06-13:
Yes, it was related to issues. 
Please try [New version](https://github.com/maddevsio/mad-location-manager-lib)

closed 2025-06-13 by Lezh1k